bitcraze/crazyradio-firmware #16: Can't make all on Mac Python 2.7 sdcc

adjavaherian, 2015-05-24:
It looks like the makefile quits after building a few bins.

[4m1r@Korgi:12:44:10 ~/Desktop/crazyradio-firmware/firmware]
$ make cradio.bin
sdcc -Iinc/ --model-large --std-sdcc99 -c src/main.c -o bin/main.rel
sdcc -Iinc/ --model-large --std-sdcc99 -c src/radio.c -o bin/radio.rel
sdcc -Iinc/ --model-large --std-sdcc99 -c src/usb.c -o bin/usb.rel
sdcc -Iinc/ --model-large --std-sdcc99 -c src/usbDescriptor.c -o bin/usbDescriptor.rel
sdcc -Iinc/ --model-large --std-sdcc99 -c src/led.c -o bin/led.rel
sdcc -Iinc/ --model-large --std-sdcc99 -c src/utils.c -o bin/utils.rel
sdcc --xram-loc 0x8000 --xram-size 2048 --model-large bin/main.rel bin/radio.rel bin/usb.rel bin/usbDescriptor.rel bin/led.rel bin/utils.rel -o bin/cradio.ihx
objcopy -I ihex bin/cradio.ihx -O binary bin/cradio.bin
make: objcopy: No such file or directory
make: *** [cradio.bin] Error 1

Phiizo, 2015-09-30:
objcopy is not supported with OS X. 
Try "brew install crosstool-ng".
In makefile, change "objcopy" to "gobjcopy".

https://pigiuz.wordpress.com/2013/09/12/resolving-symbols-conflict-between-libraries-on-osx-with-objcopy/

ataffanel, 2015-10-20:
Added instruction in readme per #20

closed 2015-10-20 by ataffanel