bitcraze/crazyradio-firmware #17: 'ftdi.h' file not found

natevw, 2015-05-28:
Running `make` in the nrfProg folder results in the following:

```
cc    -c -o nrfSpi.o nrfSpi.c
nrfSpi.c:30:10: fatal error: 'ftdi.h' file not found
#include <ftdi.h>
         ^
1 error generated.
make: *** [nrfSpi.o] Error 1
```

natevw, 2015-05-28:
I was able to compile with the following changes:

```
diff --git a/nrfProg/Makefile b/nrfProg/Makefile
index 1057c1e..878c73b 100644
--- a/nrfProg/Makefile
+++ b/nrfProg/Makefile
@@ -1,5 +1,5 @@
 
-LDFLAGS=-lftdi
+LDFLAGS=-lftdi1
 
 all: nrfProg
 
diff --git a/nrfProg/commands.c b/nrfProg/commands.c
index 92c7ecb..1cd0d65 100644
--- a/nrfProg/commands.c
+++ b/nrfProg/commands.c
@@ -136,7 +136,7 @@ int cmdProgram(int addr, char *data, int len) {
 
   if (addr>30*1024) {
     printf("ERROR: Writing above 30K is disabled!\n");
-    return;
+    return -1;
   }
 
   if((addr+len)>30*1024) {
```

and then running make with:

```
CFLAGS=-I$(brew --prefix)/include/libftdi1 LIBRARY_PATH=$(brew --prefix)/Cellar/libftdi/1.1/lib make
```

(for some reason, Homebrew's version of libftdi installs itself with the "1" suffix, and doesn't link its static library in /usr/local/lib, so it's a bit messy…but, I think have the executable I need now!)

edw1413, 2016-11-27:
what files does this go in?

ataffanel, 2016-11-27:
Can you please precise the question: Which file, in what?

edw1413, 2016-11-27:
the above code to fix the "nrfSpi.c:30:10: fatal error: 'ftdi.h' file not found" error.  in issue #17

krichardsson, 2016-11-28:
The first diff is for `nrfProg/Makefile` and the second for `nrfProg/commands.c`

closed 2022-03-16 by knmcguire