bitcraze/crazyradio-firmware #18: Trouble using nrfProg with BusBlaster on OS X

natevw, 2015-05-28:
When trying to use the BusBlaster to flash an nRF module, I would get this error:

> unable to open ftdi device: -3 (device not found)

…because the product ID is different than the script expects. This was resolved via:

```
diff --git a/nrfProg/nrfSpi.c b/nrfProg/nrfSpi.c
index 686219d..e284553 100644
--- a/nrfProg/nrfSpi.c
+++ b/nrfProg/nrfSpi.c
@@ -85,7 +85,8 @@ int spiInit()
 
   ftdi_init(&ftdic);
 
-  if((ret = ftdi_usb_open(&ftdic, 0x0403, 0xcff8)) < 0)
+  //if((ret = ftdi_usb_open(&ftdic, 0x0403, 0xcff8)) < 0)
+  if((ret = ftdi_usb_open(&ftdic, 0x0403, 0x6010)) < 0)  // BusBlaster v4 enumeration
   {
     fprintf(stderr, "unable to open ftdi device: %d (%s)\n", ret, 
             ftdi_get_error_string(&ftdic));
```

---

Fixing that leads to this error:

> unable to open ftdi device: -5 (unable to claim usb device. Make sure the default FTDI driver is not in use)

This can be resolved by running `sudo kextload -bundle com.apple.driver.AppleUSBFTDI` before using nrfProg. See <http://alvarop.com/2014/01/using-busblaster-openocd-on-osx-mavericks/> for more details on that.

knmcguire, 2022-03-09:
Hi! Sorry for the extremely late reply! And thanks for mentioning this issue. Since this repo is stil being used, I'll keep this issue open for anybody that uses the bus blaster

natevw, 2022-03-15:
Thanks! I haven't retested on any newer macOS so while the PID tip may still be helpful (and something you might update the code for?), anyone hitting this should note that the AppleUSBFTDI workaround doesn't apply. At least Apple has renamed that extension, and I found at e.g. https://github.com/eblot/pyftdi/blob/master/pyftdi/doc/troubleshooting.rst#error-access-denied-insufficient-permissions a note:

> From macOS 10.14 "Mojave", the Apple kernel extension peacefully co-exists with libusb and PyFtdi, so you no longer need - and should not attempt - to unload the kernel extension. If you still experience this error, please verify you have not installed another driver from FTDI, such as FTDI's D2XX.

(That's from an unrelated Python project but I'm assuming would also apply to the libftdi usage in this project.)