bitcraze/crazyradio-firmware #24: Scans show crazyflies on weird channels

jrjbertram, 2016-03-11:
Hi...  not sure this is a crazyradio firmware bug, but didn't see a better place to ask this question and report this issue.  If there's a better place, I can take the question there!

Whenever I scan for my crazyflie, I seem to sporadically get false returns from my crazyflie on other channels.  For example, if I configure my crazyflie for channel 80 at 250 kbps, I might see returns from 2 or 3 different channels... say channels 22, 80, and 130 (made up numbers... can't test it at the moment).  

From what I've been reading in the datasheet of the radio chip and poking through the code, wiki, and forums, that doesn't sound like expected behavior.  It seems like the radios should be able to stay on their respective channels without leaking into other frequency ranges.

Thinking in terms of EMI...  I've tested this in a couple of different locations, so I don't think it's any specific source of EMI such as a microwave or particular wifi router or device causing interference.  I only have one PC to test this with at the moment, so can't say for sure it happens with other PCs.  This also happens with the motors off, so I don't think this is EMI from the motors operating.

This is with a crazyradio PA and crazyflie 2.0 purchased from seeedstudio within the last month or so.  I've seen this behavior with multiple crazyradios and multiple crazyflies, though.

Does anyone else see this behavior?  Any ideas or explanation? 

Thanks!
- Josh.

ataffanel, 2016-03-30:
Hi,

This is a known bug. It seems to be comming from non-linearity in the radio path: the high power radio signal is leaking on other channels.

To make the problem less present you can move Crazyflie further away from the radio.

One solution would be to assign a different address for each channel, this would filter the packets coming from a wrong channel. This is a modification of the client and the Crazyflie. I will create tickets on both projects.

ataffanel, 2017-07-20:
This has no fix possible in the Crazyradio firmware, closing.

closed 2017-07-20 by ataffanel