bitcraze/crazyradio-firmware #33: Firmware update fails with Crazyradio PA, running firmware v0.1

rohern, 2016-12-15:
I get a timeout on this line in NrfUsbBootloader.py

`data = self.handle.bulkRead(0x81, 64,1000)`
`usb.core.USBError: [Errno 60] Operation timed out`

Anyone know where to start?

ataffanel, 2016-12-15:
There has never been a public release of the Crazyradio firmware V0.1. Where did you see this version?

What command are you running to get this error?

rohern, 2016-12-15:
I was playing with crazyradio.py. I am able to find the device after changing `CRADIO_PID = 0x7777` to `CRADIO_PID = 0x0101`.

This line in Crazyradio.__init__ yields 0.1:
`self.version = float("{0:x}.{1:x}".format(self.dev.bcdDevice >> 8,
                                 self.dev.bcdDevice & 0x0FF))`

dev.bcdDevice = 1

This raises the exception `"This driver requires Crazyradio firmware V0.3+"` which led me down the path to trying to update the firmware.

ataffanel, 2016-12-19:
Hehe, PID 0101 is for the Nordic Semiconductor bootloader. If you hack the Crazyradio driver to talk to something which is not a Crazyradio then you will have random results, this is not unexpected :-). Anyway this part is good, I was a bit concerned by seing Crazyradio with 0.1 version in the wild.

Have you tried flashing: https://github.com/bitcraze/crazyradio-firmware#flashing-the-firmware the latest release: https://github.com/bitcraze/crazyradio-firmware/releases/tag/0.53 ?

rohern, 2016-12-19:
I see, I went down that path this weekend. I was misinformed by the previous user about the state of the device.

When I go through the flashing steps I get the following (this is on Mac OS X Sierra):
`ryan$ python usbtools/launchBootloader.py 
Bootloader already launched.`

`ryan$ python usbtools/nrfbootload.py flash /Users/ryan/Downloads/crazyradio-firmware-0.53/cradio-pa-0.53.bin 
Found nRF24LU1 bootloader version
Traceback (most recent call last):
  File "usbtools/nrfbootload.py", line 73, in <module>
    print "Found nRF24LU1 bootloader version", bl.getVersion()
  File "/Users/ryan/Downloads/crazyradio-firmware-0.53/usbtools/NrfUsbBootloader.py", line 73, in getVersion
    data = self.handle.bulkRead(0x81, 64,1000)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/usb/legacy.py", line 165, in bulkRead
    return self.dev.read(endpoint, size, timeout)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/usb/core.py", line 988, in read
    self.__get_timeout(timeout))
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/usb/backend/libusb1.py", line 833, in bulk_read
    timeout)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/usb/backend/libusb1.py", line 936, in __read
    _check(retval)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 60] Operation timed out`

I played around with nrfdude to try to understand the source of the problem. I'm not able to even read the flash off of the device.

ataffanel, 2017-01-13:
Did it work with the raspberry pi? It might be that the bootlader is damaged. If so I might have an image of the bootloader that you could flash (if you have setup the raspi it is possible to flash a bootloader with the raspberry pi).

ataffanel, 2017-07-18:
I am assuming that this problem is solved. Closing the ticket

closed 2017-07-18 by ataffanel