bitcraze/crazyradio-firmware #34: on Mac maybe not found objcopy

mathcoder23, 2017-03-13:
on Mac:
if we used brew install binutils
README.md tell me 
`ln -s /usr/local/bin/gobjcopy /usr/local/bin/objcopy`
but maybe brew not specified to /usr/local/bin/gobjcopy of binutils 
For instance said me，
I not found /usr/local/bin/gobjcopy 
but it : /usr/local/Cellar/binutils/2.28/bin/gobjcopy
so 
`ln -s /usr/local/Cellar/binutils/2.28/bin/gobjcopy /usr/local/bin/objcopy`
propose README.md such description
```
ln -s /usr/local/bin/gobjcopy /usr/local/bin/objcopy 
(permiss termial found gobjcopy otherwise view brew install binutils location)
```

ataffanel, 2017-03-13:
Thanks for the repport. Does ```which gobjcopy``` works to find the file location, if so we could replace the documenation by
```
ln -s `which gobjcopy` /usr/local/bin/objcopy
```

krichardsson, 2017-03-13:
On my machine (El Capitan V10.11.6) I get (after ```brew install binutils```)
```
which gobjcopy
/usr/local/bin/gobjcopy

ls -la /usr/local/bin/gobjcopy
lrwxr-xr-x  1 kristoffer  admin  36 13 Mar 14:49 /usr/local/bin/gobjcopy -> ../Cellar/binutils/2.28/bin/gobjcopy
```

@mathcoder23 what do you get for ```which gobjcopy```?
It looks like the link in /usr/local/bin was not created for you?

mathcoder23, 2017-03-17:
terminal exe `which gobjcopy`   will return null;
terminal exe `gobjcopy`will return -bash: gobjcop: command not found;
my machine( Sierra 10.12),mybe brew configuration problem;on installed，Does not automatic  add to environment variable ,or Does not automatic  copy `/usr/local/Cellar/binutils/2.28/bin` to `/usr/local/bin`
look [http://stackoverflow.com/questions/29422345/installed-just-not-linked](http://stackoverflow.com/questions/29422345/installed-just-not-linked)
I used `brew doctor`
hint me unlinked to binutils ,when exe brew link bintuils ,terminal hint 
```
Linking /usr/local/Cellar/binutils/2.28... 
Error: Could not symlink share/info/bfd.info
Target /usr/local/share/info/bfd.info
is a symlink belonging to sdcc. You can unlink it:
  brew unlink sdcc

To force the link and overwrite all conflicting files:
  brew link --overwrite binutils

To list all files that would be deleted:
  brew link --overwrite --dry-run binutils
```
maybe have file conflict
so I used  `brew link --overwrite binutils`
final success brew link sdcc,binutils
terminal exe `which gobjcopy`   return /usr/local/bin/gobjcopy;
if not found gobjcopy first we need exe ` brew doctor ` check sdcc binutils link to normal,if not,can exe `brew link --overwrite binutils`,or exe `brew link --overwrite sdcc`
In this regard,if we first install binutils,maybe hint `not found sdcc` ,use  `brew link --overwrite sdcc`
so
```
ln -s `which gobjcopy` /usr/local/bin/objcopy
```
`which gobjcopy` maybe can't found gobjcopy path
`we need hint user fix not found gobjcopy of cause`

krichardsson, 2017-03-17:
Yes, this looks like a problem related to brew, some sort of conflict between sdcc and binutils. We could possibly add something to the read me to notify users that this type of problem exists, but I'm not sure. I think the forum is a good fit for this type of information as well. 

I will close this issue since it seems to be solved, or at least out of our control.

Please feel free to submit a pull request for the readme.md file to hint other users if you like :-)

Thanks!

closed 2017-03-17 by krichardsson