bitcraze/crazyradio-firmware #36: Crazyradio firmware 1.0

ataffanel, 2017-07-26:
I am currently focusing on cleaning up the Crazyradio firmware, mostly looking at closing #25 and #9. When it is done I would like to tag the firmware 1.0.

This is an RFC mostly concerning Crazyswarm. @whoenig, is there anything that still needs to be merged in order to make the firmware compatible with Crazyswarm?

whoenig, 2017-07-26:
Everything for the Crazyswarm is in it already. Perhaps related to #9: I did see many issues with USB3 (high packet drop rates I believe), and usually just plug into USB2 or use a USB2 hub if the computer has USB3 only.

ataffanel, 2017-07-27:
Thanks for the report about USB3, I created a ticket for it (#37). As far as I can tell it is not be directly related to #9 that I am currently tracking. If my assumption is correct for the USB3 problem, it cannot be solved without a breaking change to the protocol (ie. adding a mode that does not require control transfers).