bitcraze/crazyradio-firmware #44: ./tools/build/build fails - missing ftdi.h file - related to #17?

gherlein, 2017-10-26:
make: Entering directory '/home/gherlein/labs/src/crazyradio-firmware/firmware'
sdcc -Iinc/ --model-small --std-sdcc99 -DCRPA -c src/main.c -o bin/main.rel
sdcc -Iinc/ --model-small --std-sdcc99 -DCRPA -c src/radio.c -o bin/radio.rel
sdcc -Iinc/ --model-small --std-sdcc99 -DCRPA -c src/usb.c -o bin/usb.rel
sdcc -Iinc/ --model-small --std-sdcc99 -DCRPA -c src/usbDescriptor.c -o bin/usbDescriptor.rel
sdcc -Iinc/ --model-small --std-sdcc99 -DCRPA -c src/led.c -o bin/led.rel
sdcc -Iinc/ --model-small --std-sdcc99 -DCRPA -c src/utils.c -o bin/utils.rel
sdas8051 -plosgff -o bin/crtxinit.rel src/crtxinit.asm
sdcc --xram-loc 0x8000 --xram-size 2048 --model-small bin/main.rel bin/radio.rel bin/usb.rel bin/usbDescriptor.rel bin/led.rel bin/utils.rel bin/crtxinit.rel -o bin/cradio.ihx
objcopy -I ihex bin/cradio.ihx -O binary bin/cradio.bin
Crazyradio PA build
make: Leaving directory '/home/gherlein/labs/src/crazyradio-firmware/firmware'
make: Entering directory '/home/gherlein/labs/src/crazyradio-firmware/nrfProg'
cc    -c -o nrfProg.o nrfProg.c
cc    -c -o nrfSpi.o nrfSpi.c
nrfSpi.c:30:18: fatal error: ftdi.h: No such file or directory
 #include <ftdi.h>
                  ^
compilation terminated.
<builtin>: recipe for target 'nrfSpi.o' failed
make: *** [nrfSpi.o] Error 1
make: Leaving directory '/home/gherlein/labs/src/crazyradio-firmware/nrfProg'

gherlein, 2017-10-26:
sudo apt-get install libftdi-dev
<snip, but normal installation logs>

gherlein@mars:~/labs/src/crazyradio-firmware$ ./tools/build/build
make: Entering directory '/home/gherlein/labs/src/crazyradio-firmware/firmware'
Crazyradio PA build
make: Leaving directory '/home/gherlein/labs/src/crazyradio-firmware/firmware'
make: Entering directory '/home/gherlein/labs/src/crazyradio-firmware/nrfProg'
cc    -c -o nrfSpi.o nrfSpi.c
cc    -c -o commands.o commands.c
commands.c: In function ‘cmdProgram’:
commands.c:139:5: warning: ‘return’ with no value, in function returning non-void
     return;
     ^~~~~~
commands.c:129:5: note: declared here
 int cmdProgram(int addr, char *data, int len) {
     ^~~~~~~~~~
commands.c:158:5: warning: implicit declaration of function ‘cmdErasePage’ [-Wimplicit-function-declaration]
     cmdErasePage(i);
     ^~~~~~~~~~~~
cc -o nrfProg nrfProg.o nrfSpi.o commands.o -lftdi
make: Leaving directory '/home/gherlein/labs/src/crazyradio-firmware/nrfProg'

I'll look to see if it's me and if not I'll submit a PR to fix this.

gherlein, 2017-10-26:
submitted PR https://github.com/bitcraze/crazyradio-firmware/pull/45

ataffanel, 2017-10-27:
Thanks, I merged your PR.

closed 2017-10-27 by ataffanel