bitcraze/crazyradio-firmware #50: [Errno 110] Operation timed out

ThomasDavidByrne, 2018-04-17:
When you do this step - python ../usbtools/launchBootloader.py 
I get this error;
Launch bootloader Traceback (most recent call last):
  File "../usbtools/launchBootloader.py", line 95, in <module>
    sys.exit(launchBootloader(True))
  File "../usbtools/launchBootloader.py", line 61, in launchBootloader
    handle.controlMsg(0x40, 0xFF, (), value=0, index=0, timeout=100)
  File "/usr/lib/python2.7/dist-packages/usb/legacy.py", line 205, in controlMsg
    timeout = timeout)
  File "/usr/lib/python2.7/dist-packages/usb/core.py", line 971, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/usr/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 819, in ctrl_transfer
    timeout))
  File "/usr/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 552, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out

Any help on fixing this issue ?

ataffanel, 2018-04-17:
- Are both LED blinking once when you plug the radio to the computer?
 - Can you share the output of ```lsusb -v -d 1915:7777 | grep bcdDevice``` (to get the Crazyradio version)?
 - Have you tried to run the command with 'sudo'?

ThomasDavidByrne, 2018-04-17:
@ataffanel  The lights come on and then off again. I think we figured out the issue is to do with my usb ports themselves as my colleague was able to run it on his just fine. I think it was disconnecting and reconnecting.

krichardsson, 2018-06-25:
Looks like this issue is sorted? I'll close it, please reopen if not.

closed 2018-06-25 by krichardsson

NicksonYap, 2018-12-17:
Fixed it! The timeout was too short!

```
  #Send the command to arm the bootloader
  handle.controlMsg(0x40, 0xFF, (), value=0, index=0, timeout=3000)
```

[image]