bitcraze/crazyradio-firmware #52: A bytes-like object is required, not 'tuple'

jalalirs, 2018-10-22:
Hello,

I am getting the below error when I run 
`sudo python3.6 ../usbtools/nrfbootload.py flash bin/cradio.bin`

Found nRF24LU1 bootloader version 18.0
Flashing:
  Flashing 7759 bytes...
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/usb/_interop.py", line 92, in as_array
    return array.array('B', data)
TypeError: integer argument expected, got float

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "../usbtools/nrfbootload.py", line 135, in <module>
    if bl.write(fileData):
  File "/Users/jalalirs/Documents/technincal_material/code/github/crazyfly/crazyradio-firmware/usbtools/NrfUsbBootloader.py", line 139, in write
    if not self.writePage(data[i:(i+PAGE_SIZE)], (address+i)/PAGE_SIZE):
  File "/Users/jalalirs/Documents/technincal_material/code/github/crazyfly/crazyradio-firmware/usbtools/NrfUsbBootloader.py", line 107, in writePage
    self.handle.bulkWrite(0x01, (CMD_PAGE_ERASE, pageNum), 1000)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/usb/legacy.py", line 154, in bulkWrite
    return self.dev.write(endpoint, buffer, timeout)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/usb/core.py", line 947, in write
    _interop.as_array(data),
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/usb/_interop.py", line 97, in as_array
    a.fromstring(data) # deprecated since 3.2
TypeError: a bytes-like object is required, not 'tuple'

ataffanel, 2018-10-23:
Hi, thanks for the report! This seems to come from pyusb that has changed its API, could you  check and tell us the version of pyusb you are using, you can do it with the command:
```
$ pip3 show pyusb
---
Metadata-Version: 2.0
Name: pyusb
Version: 1.0.2
Summary: Python USB access module
(...)
```

jasperSpeicher, 2018-11-29:
I have this issue and I have V 1.0.2. What version should I have?

ataffanel, 2018-11-29:
@jasperSpeicher on what system do you see the problem and what version of Python are you running?

aaarghhh, 2019-02-27:
Hi @ataffanel ,  same here : 

Debian
Python 3.5.3
pyusb 1.0.0

Sundar0989, 2019-04-24:
I am also facing the same issue here.
Name: pyusb
Version: 1.0.2

I have Mac OS Sierra version 10.12.6.

Please assist.

aaarghhh, 2019-04-25:
@Sundar0989 I don't know if I can help, for my purphose I used this one -> https://github.com/BastilleResearch/nrf-research-firmware

ataffanel, 2019-04-30:
The problem seems to be Python3: the nrfbootload.py script is not compatible with Python3. This should be fixed.

sheevamsharma, 2020-06-30:
Hey, has this issue been solved yet for Python3? I am now also running into the same error

knmcguire, 2020-07-01:
We unfortunately have not yet I'm afraid... sorry about that. I do know that @tobbeanton was able to flash the crazyradio on ubuntu 20.04 with an installed python2. 

Is your issue that your crazyradio stopped working after flashing?

kripton, 2021-04-04:
Hi, I just wanted to note that I am also unable to flash my CrazyRadio PA since the `nrfbootload.py` doesn't work with Python 3 :(
Also running the `2to3` automatic conversion script was not able to fix it properly

ataffanel, 2021-04-13:
I converted nrfbootload.py to Python3, it does work with Python 3.8 (but should not work anymore with Python2 which should not be a problem anymore ...).

commit 129d2939 (2021-04-13): Fixes #52: Convert nrfbootload to Python3
closed 2021-04-13 by ataffanel