bitcraze/crazyradio-firmware #57: Can only connect to CF2 through USB or in bootloader mode

davidoort, 2020-03-15:
I am running the following:
Crazyflie 2.0: I have flashed `firmware-cf2-2020.02.zip` through the `cfclient` GUI.
Crazyradio PA: I have flashed `cradio-pa-0.53.bin` by running the following commands from the root of this repository:
```
make CRPA=1
```

krichardsson, 2020-03-25:
Your Crazyflie might be on a different address.

In the python client:
1. Connect your Crazyflie with USB
1. Go to the Connect/Configure 2.X menu 
1. Check which address it is set to

You can either enter this address in the client (Address field) and th CF should be found when scanning, or you can change the address in the Crazyflie to the the default address 0xE7E7E7E7E7 (you must restart the CF for the new address to be active)

davidoort, 2020-03-26:
This is what I see when I open the Python Client GUI
[image]
When I hover over Connect, the Configure 2.X option is not clickable. What could be causing this?

krichardsson, 2020-03-26:
Ok, you need to click the "Connect" button first to connect to the Crazyflie (it will connect via USB). After that the Connect/Configure 2.X menu should be enabled

davidoort, 2020-03-27:
Great! I connected through USB, changed the address from what it was set (0xE7E7E7E735) to 0xE7E7E7E7E7 and after writing and restarting the CF, I was able to scan the CF through the radio. Thank you so much!

closed 2020-03-27 by davidoort