bitcraze/crazyradio-firmware #62: flashing issue

antbono, 2022-12-15:
Hello everyone! 
I have just followed the instructions to compile the firmware on a freshly installed ubuntu 22.04 (my default python version is 3.8.10). I can start the bootlader but when I try to flash, I obtain the following error. 

[image]

Please consider that I successfully use the dongle with the crazyflie python client for ubuntu. 

Thank you for your attention!

kripton, 2022-12-15:
Hi @antbono : Do you have the package "pyusb" installed? https://pyusb.github.io/pyusb/ https://pypi.org/project/pyusb/

If not, try installing it and try again

antbono, 2022-12-15:
Hi @kripton!

Yes I have installed that library

[image]

antbono, 2022-12-15:
Sorry I missed 'sudo' in the pip install command...

krichardsson, 2023-01-10:
As far as I understand this issue has been fixed and I will close this issue. Please reopen if there is still a problem!

closed 2023-01-10 by krichardsson