bitcraze/crazyradio-firmware #63: Crazyradio PA not recognized by Raspberry Pi

willgraham345, 2023-08-16:
I'm unable to get a Crazyradio to be recognized by a raspberry Pi as anything other than a "Nordic Semiconductor ASA". My output from `lsusb` is shown below:
```
ubuntu@ubuntu:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0483:5740 STMicroelectronics Virtual COM Port
Bus 001 Device 004: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
Bus 001 Device 003: ID 1915:7777 Nordic Semiconductor ASA
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

I've also changed the `/etc/udev/rules.d/99-bitcraze.rules` to contain:
```
# Crazyradio (normal operation)
SUBSYSTEM=="usb", ATTRS{idVendor}=="1915", ATTRS{idProduct}=="7777", MODE="0664", GROUP="plugdev"
# Bootloader
SUBSYSTEM=="usb", ATTRS{idVendor}=="1915", ATTRS{idProduct}=="0101", MODE="0664", GROUP="plugdev"
# Crazyflie (over USB)
SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5740", MODE="0664", GROUP="plugdev"
```

I've tried to build the Crazyradio on the Raspberry Pi and I've run into similar issues (I unplugged, and replugged the radio between the two `lsusb` commands)
```
ubuntu@ubuntu:~/crazyradio-firmware/firmware$ make CRPA=1
Crazyradio PA build
ubuntu@ubuntu:~/crazyradio-firmware/firmware$ python3 ../usbtools/launchBootloader.py
Launch bootloader .
Bootloader started
ubuntu@ubuntu:~/crazyradio-firmware/firmware$ python3 ../usbtools/nrfbootload.py flash bin/cradio.bin
Found nRF24LU1 bootloader version 18.0
Flashing:
  Flashing 7861 bytes...
Flashing done!
Verifying:
  Reading bin/cradio.bin...
  Reading 7861 bytes from the flash...
Verification succeded!
ubuntu@ubuntu:~/crazyradio-firmware/firmware$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0483:5740 STMicroelectronics Virtual COM Port
Bus 001 Device 004: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
Bus 001 Device 007: ID 1915:0101 Nordic Semiconductor ASA
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
ubuntu@ubuntu:~/crazyradio-firmware/firmware$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0483:5740 STMicroelectronics Virtual COM Port
Bus 001 Device 004: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
Bus 001 Device 008: ID 1915:7777 Nordic Semiconductor ASA
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

When I plug the same Crazyradio back into my linux machine, I can see `Nordic Semiconductor ASA Crazyradio PA USB Dongle". The radio performs great, and I can successfully fly and control Crazyflies from that machine. 

Is there a reason as to why I'm unable to use this as a radio with these settings?

ataffanel, 2023-08-18:
Hi,

Lsusb showing the Crazyradio as an "Nordic Semiconductor" device is correct: Crazyradio PA is using the USB Vendor ID from Nordic, the chip manufacturer Crazyradio is based on. Future Crazyradio 2.0 firmware will use our VID and will appear as "Bitcraze AB".

So this is not a problem and if anything it proves that the Crazyradio is well recognized by your Raspberry PI.

You mentioned that the radio works well on PC, what does not work on the raspberry pi?

willgraham345, 2023-08-21:
The issue comes when I try to fly the crazyflie with example scripts (i.e. multiranger_push.py and other) do not interact with the crazyflie at all. This seems to be a bigger issue, as my client is no longer pairing with the crazyflie. I correctly see a radio option drop down on the scan, but cannot connect anymore. I have the usb permissions set up correctly, but keep running into issues with any communication between the crazyflie and the client.

ataffanel, 2023-08-28:
If the radio is well detected by the Host, you should be able to observe 2 things:
 - When connecting the radio, both RED and GREEN LEDs will blink once
 - When clicking the scann button in the client, the RED LED should light UP and, if a Crazyflie is switched-on and on the right address, the GREEN LED will blink very briefly

Are you able to see both behavior, on PC and on the raspberrypi?

willgraham345, 2023-08-29:
I was able to get the crazyradio working with my PC and had no issues flying the crazyflie around. The issue happens when I try and fly the crazyflie with commands from the turtlebot. 

Unfortunately, we can't use the client with the pi, as we are running it in a semi-headless fashion. I've been using the [example files](https://github.com/bitcraze/crazyflie-lib-python/tree/master/examples/step-by-step) in the crazyflie python firmware. I played around with the radio examples as well, and saw that the radio is being registered. When I use the [scan.py](https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/radio/scan.py) script tool I see output:
```bash
Scanning interfaces for Crazyflies...
Crazyflies found:
radio://0/80/2M
```

When I connect the radio, both RED and GREEN LEDs blink one time. When I connect the radio to the pi with a Crazyflie switched-on, I don't see the GREEN LED. I'm using default addresses for the Crazyflie (radio://0/80/2M).

willgraham345, 2023-09-07:
After quite a bit of trial and error, it became clear that the flow deck was broken and causing the issue. When I reattached a new flow deck everything worked perfectly. Thank you for the support through the debugging!!

closed 2023-09-07 by willgraham345