bitcraze/crazyradio-firmware #65: Fix bootloader on Orange Pi is supported?

GermanAizek, 2023-12-13:
Can you tell me if there is fix bootloader for GPIO Orange Pi?
I do not know if Raspberry and Orange GPIO ports match, but I doubt it very much.

ataffanel, 2023-12-14:
It should work if you port the script to the Orange PI. Most SBC try to have raspberry-pi compatible GPIO port so while I do not know how to do it, I do not see any reason why it would not work. The first step would be to find how to access GPIO from python on an OrangePI

GermanAizek, 2023-12-15:
@ataffanel I can do porting fix bootloader script for OrangePi, pinout is same, but orangepi has a maximum 26 pins.

GermanAizek, 2023-12-20:
@ataffanel script also works for OrangePi, it is enough to replace line `import RPi.GPIO as GPIO` to `import OPi.GPIO as GPIO`.
OPi.GPIO library has same behavior, functions and errors as RPi.GPIO.
But problem is that I have `[Error] device is busy`, on RESET port 3 pin. Do I need to connect Crazyradio to USB while running script? LED was not blinking/enabled crazyradio when I plugged it into USB.

ataffanel, 2023-12-21:
Great that it was that easy for the coding part! I was expecting other SBC to try to be as compatible as possible with Raspberrypi.

The board must be powered somehow, so if you are not connecting VCC from the expansion port (it does not seems to be the case in the script), then yes you must connect it to the USB port. Otherwise the board will not receive any power and the CPU cannot start.

As for the LED, I assume that the reason you need to run this script is that you need to recover the board, so it is normal that nothing is blinking. If successful, the board will jump into the USB bootloader that will allow you to flash an official firmware.

GermanAizek, 2023-12-24:
@ataffanel 
```
ubuntu@ubuntu-NB:~/GIT/crazyradio-firmware/firmware$ make CRPA=1
Crazyradio PA build
ubuntu@ubuntu-NB:~/GIT/crazyradio-firmware/firmware$ python3 ../usbtools/launchBootloader.py 
Error!, cannot find the CrazyRadio USB dongle!
```

firmware makefile used python2, I replaced it with python 3, but build did not started, as well as launch bootloader does not see crazyradio pa.

```
ubuntu@ubuntu-NB:~/GIT/crazyradio-firmware/firmware$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 09da:2403 A4Tech Co., Ltd. 2.4G Device
Bus 003 Device 002: ID 0408:a060 Quanta Computer, Inc. HD Webcam
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 04ca:3016 Lite-On Technology Corp. 
Bus 001 Device 002: ID 03f0:0024 HP, Inc KU-0316 Keyboard
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

knmcguire, 2024-01-02:
Hi! Sorry for the late reply. @ataffanel should be back next week and should be able to help better than me.

Are you using the crazyradio 2 and not the crazyradio 1 by any chance? Just to make sure since I've seen this happen before.

GermanAizek, 2024-01-06:
@knmcguire yes I am using Crazyradio 2 PA latest revision board.

knmcguire, 2024-01-08:
aha that might be the problem. These instructions do not apply at all for Crazyradio 2.0, only for the crazyradio PA (so the first version). I'll close this issue as this repository is only related to the crazyradio version 1. This is the repository for crazyradio 2: https://github.com/bitcraze/crazyradio2-firmware

But before going into that development documentation, you should follow these instructions first: https://www.bitcraze.io/documentation/tutorials/getting-started-with-crazyradio-2-0/, but I hope you haven't broken anything that doesn't work anymore.

If you are still having issues, could you please start a thread on https://discussions.bitcraze.io/. that is the appropriate place for support.

knmcguire, 2024-01-26:
reopening since this discussion mentioned it is actually a crazyradio PA. https://github.com/orgs/bitcraze/discussions/1152

ataffanel, 2024-03-19:
Closing this issue since there is not much we can do on it in the project itself. Feel free to discuss on the discussion if you need more help.

closed 2024-03-19 by ataffanel