bitcraze/crazyradio-firmware #66: Can't flash the content by spi on raspberry

duguguang, 2024-01-15:
refer to https://www.hackster.io/ataffanel/fixing-crazyradio-usb-bootloader-with-a-raspberry-pi-08796b, And i have checked the pin.

sudo python3 fix_bootloader_raspi.py
Erasing page 0...
Writing 'JL 0x7800' instruction...
Flash content:
ff
ff
ff
Fix written to flash. Disconnect the PROG wire, reconnect Crazyradio

knmcguire, 2024-01-15:
Hi! Are you sure that you are using the Crazyradio PA (version 1.0) and **not** Crazyradio 2.0 for these instructions. These operate in very different ways.

duguguang, 2024-01-15:
Yes, i am using the pa1.0

ataffanel, 2024-01-16:
It seems that you can erase the flash but not write it. It can also be a problem of GPIO of cabling.

What version of raspberry pi are you using? Is the LED on the Crazyradio lights up during the process?

Can you connect MISO to a ground on the raspi instead of to the Crazyradio and see if you read back 0x00 instead of 0xFF? This will verify that this cable is working well.