bitcraze/crazyradio-firmware #67: Input/Output Error

JCdeception, 2024-01-30:
python2 ../usbtools/launchBootloader.py
Launch bootloader Traceback (most recent call last):
  File "../usbtools/launchBootloader.py", line 95, in <module>
    sys.exit(launchBootloader(True))
  File "../usbtools/launchBootloader.py", line 61, in launchBootloader
    handle.controlMsg(0x40, 0xFF, (), value=0, index=0, timeout=5000)
  File "/usr/lib/python2.7/dist-packages/usb/legacy.py", line 211, in controlMsg
    timeout = timeout)
  File "/usr/lib/python2.7/dist-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/usr/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/usr/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 5] Input/Output Error
any help on fixing this issue

HoAd-sc, 2024-02-21:
> python2 ../usbtools/launchBootloader.py Launch bootloader Traceback (most recent call last): File "../usbtools/launchBootloader.py", line 95, in sys.exit(launchBootloader(True)) File "../usbtools/launchBootloader.py", line 61, in launchBootloader handle.controlMsg(0x40, 0xFF, (), value=0, index=0, timeout=5000) File "/usr/lib/python2.7/dist-packages/usb/legacy.py", line 211, in controlMsg timeout = timeout) File "/usr/lib/python2.7/dist-packages/usb/core.py", line 1043, in ctrl_transfer self.__get_timeout(timeout)) File "/usr/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer timeout)) File "/usr/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 595, in _check raise USBError(_strerror(ret), ret, _libusb_errno[ret]) usb.core.USBError: [Errno 5] Input/Output Error any help on fixing this issue

Solved it?

HoAd-sc, 2024-02-21:
Seems like I'm using Crazyradio 2.0? What's the solution?

gemenerik, 2024-02-22:
> I'm also having the same issue, I'm inside kali, and while performing the launchBootloader.py

Do you get exactly the same error? If so, can you give some context like @ataffanel suggested? 

> I use lsusb to discover the 1915:7777 Crazyradio device

The device showing in lsusb does not imply _access_ to the USB device.

HoAd-sc, 2024-02-22:
I found the reason: I am using crazyradio2, and this project is crazyradio's

ataffanel, 2024-02-23:
Yes this is the reason. Crazyradio 2 is very different and is using a proper safe bootloader and the same debug port than the Crazyflie. So if you want to start working with the Crazyradio2 firmware, you can find our Zephyr-based firmware there: https://github.com/bitcraze/crazyradio2-firmware

Please note that Crazyradio 2 already works with Crazyswarm (2) out of the box so there is not need to flash it for that reason.

closed 2024-02-23 by ataffanel